shaise/FreeCAD_SheetMetal #431: Problem I generate an object with skeetmetal, u-shape,

ik4yaz, 2025-01-03:
### Describe your issue

Problem description

Problem
I generate an object with skeetmetal, u-shape,
I use unfold and it generates the flat sketch, it's ok, in techdraw I build the 2d table, it's ok.
but if I then want to use the object in a2plus it only inserts the flat and solid shape, not the "u".
Full version info

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods:
 * A2plus 0.4.68
 * dxf-library
 * fasteners 0.5.34
* freecad.gears 1.3.0
* Movie 2024.2.15
* Render 2024.12.15
* sheetmetal 0.6.13
P.s. general problem! the unfold tool does not work correctly, try to generate a sketch inserter after step in sheetmetal set thickness and radius generate shape, then I lift one side generate a fold, but then unfold does not create flat sketch only 3d flat shape.
Fabrizio 
mail: garuti.fabrizio@libero.it

ik4yaz, 2025-01-04:
[FreeCAD 1.0.0 2025-01-04 09-45-43-p.webm](https://github.com/user-attachments/assets/e0710e18-4103-43a9-a97f-6c401c9ec33d)

It's easier to understand from the video

shaise, 2025-01-06:
I dont think it is a sheetmetal issue, I think A2plus always take the tip of a body. (Try to do it with a body that have only part design features and let me know if you can use features that are not the last one in the body)
I am thinking however to have an option for the unfold to be created outside the body as a separate object (similar to what it was before but still parametric) This might solve your problem.

ik4yaz, 2025-01-06:
Hi, I have done various tests, but the result does not change in a2plus, while everything is normal with assembly, however I confirm that in version 0.20.2 everything worked well even in previous versions, this unfold is an important thing to give the techdraw drawings to those who have to work the sheet metal, but then it must also work well in assembly, thanks for your great work.

luzpaz, 2025-02-01:
>  I am thinking however to have an option for the unfold to be created outside the body as a separate object (similar to what it was before but still parametric) This might solve your problem.

Rename the ticket to this ?

shaise, 2025-02-02:
The reason it worked well on previous versions is because the unfold was a separate object. I will add a checkbox that lets the unfold object be a separate object (still parametric but outside the body) this way it will work with assembly.