shaise/FreeCAD_SheetMetal #441: NewUnfolder, parts without bends do not unfold

GS90, 2025-01-13:
### Describe your issue

Yes, I understand that it is not entirely correct to unfold a flat part... but such parts are often found in assemblies along with those with bends, and I think they should be processed in a single process.

When trying to unfold a flat part, an error occurs:
```
File "/home/***/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalUnfoldCmd.py", line 260, in execute
  shape, sketches = self.newUnfolder(fp)
File "/home/***/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalUnfoldCmd.py", line 209, in newUnfolder
  sel_face, unfolded_shape, bend_lines, root_normal = SheetMetalNewUnfolder.getUnfold(
File "/home/***/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalNewUnfolder.py", line 1108, in getUnfold
  unfolded_shape, bend_lines = unfold(shp, root_face_index, bac)
File "/home/***/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalNewUnfolder.py", line 995, in unfold
    graph_of_sheet_faces = build_graph_of_tangent_faces(shape, root_face_index)
File "/home/***/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalNewUnfolder.py", line 794, in build_graph_of_tangent_faces
  raise RuntimeError(errmsg)
<class 'RuntimeError'>: No faces were found that are tangent to the selected face. Try selecting a different face, and/or confirm that the shape is a watertight solid.
Recompute failed!
```

The old algorithm allowed flat parts to be unfolded...

### FreeCAD version info + SheetMetal WB version

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.5.35
  * Assembly4 0.50.15
  * Plot 2022.4.17
  * CfdOF 1.29.1
  * sheetmetal 0.7.1
  * addFC 1.0.8
```

### Put here any screenshots or videos (optional)

[Amplifier.zip](https://github.com/user-attachments/files/18402774/Amplifier.zip)

alexneufeld, 2025-01-13:
please test:  https://github.com/shaise/FreeCAD_SheetMetal/pull/442

You can also assign me to anything related to the new unfolder :+1:

GS90, 2025-01-14:
> please test: #442

Great, it works.
Thanks!

PR #442 (2025-01-14): allow unfolding flat shapes
closes #441
closed 2025-01-14 by shaise