shaise/FreeCAD_SheetMetal #443: Name `nx` is not defined

pierreporte, 2025-01-14:
### Describe your issue

1. Open FreeCAD
2. Create a Parametric Part
3. Open the Sheet Metal workbench
4. An error dialog box appears: “Name ‘nx’ is not defined” and the report view contains the trace below.
5. The toolbar stays as if it was still Part Design. If you switch to an other workbench and back to Sheet Metal, its toolbar would be empty of sheet metal tools (only standards tools are there).

```
19:43:49  NOTICE: Addon "/home/thomas/.local/share/FreeCAD/Mod/sheetmetal.backup1731333427.8204525" disabled by presence of ADDON_DISABLED stopfile
19:44:08  name 'nx' is not defined
19:44:08  Traceback (most recent call last):
  File "/home/thomas/.local/share/FreeCAD/Mod/sheetmetal/./InitGui.py", line 68, in Initialize
    import SheetMetalUnfoldCmd
  File "/tmp/.mount_FreeCAPilFAf/usr/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/thomas/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalUnfoldCmd.py", line 40, in <module>
    import SheetMetalNewUnfolder
  File "/tmp/.mount_FreeCAPilFAf/usr/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/thomas/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalNewUnfolder.py", line 760, in <module>
    def build_graph_of_tangent_faces(shp: Part.Shape, root: int) -> nx.Graph:
                                                                    ^^
```

### FreeCAD version info + SheetMetal WB version

```shell
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39841 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 7a5a3d1ffcd6a5a01c90f6d17f5dd7eb8f6fa7aa
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CosmeticThread3D
  * sheetmetal.backup1731333427.8204525 0.5.3 (Disabled)
  * Curves 0.6.53
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.1
```

shaise, 2025-01-14:
@pierreporte , this is not exactly a bug, but rather a limitation of the addon manager.
You need to uninstall sheetmetal, refresh the addon manager cache then reinstall it, then it will prompt you to install the missing library dependency. 
I need to figure a way to prompt that.

pierreporte, 2025-01-14:
Indeed. I think it’s more an issue with the add-on manager. It should notify the user that an add-on has a new dependency as long as the add-on declares it.

g-radam, 2025-01-15:
Just got hung up on this, glad to see this issue exists :)

shaise, 2025-01-15:
I have improved the networkx detection system. It will now show the above quick fix message, but will not disable the WB (just revert to the old unfolder)