shaise/FreeCAD_SheetMetal #444: K-Factor radio buttons have no effect on KFactor Standard property

FC-FBXL5, 2025-01-14:
Version 0.7.01:

While updating the Unfold wiki page I tried to change the K-Factor standard from ANSI to DIN in the Unfold sheet metal object Task panel.
I can switch the radio buttons but neither using the Update button nor pressing the OK button changes the KFactor Standard property.
If I change the property in the property window the right radio button is active the next time the task panel is opened.

For the K-Factor value and KFactor property it is the other way around. It also doesn't accept values greater than 1 for DIN factors - range is from 0 to 2.

Unfolding itself seems to work more reliably and faster now with the new unfolded.

shaise, 2025-01-15:
Thanks! I will take a look.

shaise, 2025-01-15:
@FC-FBXL5 , should be fixed now, please test.

FC-FBXL5, 2025-01-17:
The radio buttons are working well now, and also conversion between ANSI and DIN values.

But whenever I open the Task panel and DIN is active the KFactor value is reset to 1 in the spin box and not set to the stored value. It is then impossible to enter a value greater than 1.

Workaround is: Switch to ANSI enter a value greater than 0.5 and switch back to DIN, now it is possible to enter values up to 2.

shaise, 2025-01-17:
Sorry about that, I always fall to this gui chicken and egg trap. Please try again.

FC-FBXL5, 2025-01-19:
I can confirm that radio buttons and KFactor value work properly now.

shaise, 2025-01-19:
Great! thanks.

closed 2025-01-19 by shaise