shaise/FreeCAD_SheetMetal #448: Error: Networkx dependency is missing and required for the new Unfolder,

ForrestErickson, 2025-01-17:
### Describe your issue

# Summary
Error message:

> 09:19:56  Networkx dependency is missing and required for the new Unfolder
> Try uninstalling SheetMetal, refresh Addon Manager's cache, and reinstall
> 

## System Setup, Reproduction
I have installed FreeCAD 1.0
Windows 10 computer.

When I open FreeCAD and start a new file and then select the sheetmetal work bench I get the following error
[image]

## Workbench version
I appear to be running the latest version **HOWEVER the version is not reported in the addon Manager.** which seam like a bug too.
[image]

### FreeCAD version info + SheetMetal WB version

```shell

```

ForrestErickson, 2025-01-17:
Notes on trying the uninstall.

[image]

Refresh cach
[image]

Select to install

[image]

[image]

[image]

[image]

Then opened a new document.
Switched to Sheetmetal work bench and no error.

[image]

ForrestErickson, 2025-01-17:
I think this works as intended so closing.

closed 2025-01-17 by ForrestErickson

shaise, 2025-01-17:
Sorry about that, this is an addon manager bug, so the whole uninstall - install is to overcome it.