shaise/FreeCAD_SheetMetal #449: invalid syntax (SheetMetalNewUnfolder.py, line 384)

ForrestErickson, 2025-01-17:
### Describe your issue

# Summary
Error messages:
09:38:45  invalid syntax (SheetMetalNewUnfolder.py, line 384)
09:38:45  Traceback (most recent call last):
  File "<string>", line 68, in Initialize
  File "C:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Users\Admin\AppData\Roaming\FreeCAD\Mod\sheetmetal\.\SheetMetalUnfoldCmd.py", line 40, in <module>
    import SheetMetalNewUnfolder
  File "C:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)

## System Setup, Reproduction
I have installed FreeCAD 0.21
Windows 10 computer.
I have the up to date work bench.
I open this file:
[ThreeU_Chassis_20250117_0944.zip](https://github.com/user-attachments/files/18455995/ThreeU_Chassis_20250117_0944.zip)

I get this error when I try to refresh the design
[image]

## FreeCad Version Note:
This file also causes FreeCAD 1.0 to produce a message about migration which I am not able to clear and so I am falling back to FreeCAD 0.21

### FreeCAD version info + SheetMetal WB version

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.68
  * Assembly4 0.50.15
  * Curves 0.6.53
  * fasteners 0.5.35
  * kicadStepUpMod 11.5.2
  * sheetmetal 0.7.3
[/code]
```

### Put here any screenshots or videos (optional)

[ThreeU_Chassis_20250117_0944.zip](https://github.com/user-attachments/files/18456047/ThreeU_Chassis_20250117_0944.zip)

ForrestErickson, 2025-01-17:
Note the sketch to which fold is applied has moved in the tree view.
[image]

The FoldRear used to be a feature added to the Sketch021. 
Now they do not have the old relationship.

Syres916, 2025-01-17:
`match` was introduced in Python 3.10 hence the invalid syntax error, I think an `if` condition is required here.

shaise, 2025-01-17:
@Syres916 , I have a release version of freecad 21, and it comes with python 3.10. How come?

```
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
```

Syres916, 2025-01-18:
@shaise I would say 90% of Windows users who downloaded 0.21.x have the _LibPack_ based build (Python 3.8.x) using the [NSIS Installer](https://github.com/FreeCAD/FreeCAD/releases/download/0.21.2/FreeCAD-0.21.2-WIN-x64-installer-1.exe) and yours is a _Conda_ based one.

shaise, 2025-01-18:
Ok than, I have changed all match statements to if-then-else. @ForrestErickson , can you see if it works?

ForrestErickson, 2025-01-18:
Regarding, ". @ForrestErickson , can you see if it works?"

I am not sure what you are asking. Is there an new version and will it load from the FreeCAD manager or do I need another method to install the workbench?  If so I need an explanation plesae.  
And is there a way to see the FreeCAD_SheetMetal version?

## My Broken Install
Another news. 
My install of the FreeCAD_SheetMetal work bench version from the day before yesterday (I think) for FreeCAD 1.0 has broken my FreeCAD 0.21. I can no longer edit my file in FreeCAD 0.21 AND I can not edit the file in FreeCAD 1.0 due to another migration bug which may or may not be related to the Sheetmetal work bench.   So I am twice blocked right now.

shaise, 2025-01-18:
> do I need another method to install the workbench? If so I need an explanation please.

No formal new version but if you uninstall and reinstall the workbench using the addon manager, you will get the updated code.

ForrestErickson, 2025-01-19:
## Notes on Installing.
[image]

[image]

I restarted FreeCAD.

## Notes on continued error
I opened the file: ThreeU_Chassis_20250117_0944.FCStd

I got the following error messages:
> 04:09:36  ThreeU_Chassis_20250117_0944#Sketch013: The Sketch has partially redundant constraints!
> 04:09:36  PartDesign::Body: Link(s) to object(s) 'Body002' go out of the allowed scope 'Body004'. Instead, the linked object(s) reside within 'Part002'.
> 04:09:36  PartDesign::Body: Link(s) to object(s) 'Body001' go out of the allowed scope 'Body005'. Instead, the linked object(s) reside within 'Part001'.

In addition the four folds and sketches are not properly associated (Fold Rear, Fold Left, Fold Right, Fold Front )
[image]
Fold Rear should be applied to Sketch021 for example.

## Bottom Line
My sketch is still broken. I am also unable to migrate this sketch to FreeCAD V1.0

shaise, 2025-01-19:
Redundant constrains of sketches are beyond the SheetMetal scope.

ForrestErickson, 2025-01-19:
Regarding, "Redundant constrains of sketches are beyond the SheetMetal scope."
This message did not appear until I did two things. Migrate to FreeCAD 1.0 and update the sheetmetal workbench.

I think the fold becoming uncoupled from its sketch happened with the move to the new sheetmetal work bench version.  Even when I try to open in version 0.21 it is uncoupled. 

How might I change my version 0.21 FreeCAD to an older sheetmetal work bench to try to confirm?

Are you able to open the file [ThreeU_Chassis_20250117_0944.zip](https://github.com/user-attachments/files/18456047/ThreeU_Chassis_20250117_0944.zip) I provided with out the error?

shaise, 2025-01-19:
Unfortunately addon manager does not let you change the version of SheetMetal, however it is possible to do it manually, though a bit complicated.
You can install TortoiseGit + git from here: https://tortoisegit.org/
then go to the installation location of the workbenches (on windows usually c:\users\<user name>\AppData\Roaming\FreeCAD\Mod), right click on the sheetmatel folder, choose tortoisegit -> swich/checkout, and in the dialog that comes out click Tag and select the version you want.
If this is too complicated, let me know what version you want and I can zip and attach it here. You will have to delete the sheetmetal folder mentioned above and replace it with the one I give you.
I will check the attached file as well and let you know.

ForrestErickson, 2025-01-19:
Regarding, "If this is too complicated, let me know what version you want and I can zip and attach it here"  

I do not know how to identify the old version (I do not even know how to see the current version on my FreeCAD 1.0). However I made the file last July. 

Could you make a guess at a version from about last July and zip up and post please?

Mean while I will ask on the Facebook page where some users are running two different version of FreeCAD with independent settings and I think addons to see if I can get my 0.21 independent of my 1.0. 

And by the way THANKS!

shaise, 2025-01-19:
Ok I tested your file on freecad 0.21, both the version with python 3.8 and the one with python 3.10. They both load the file with no error using the latest SheetMetal

ForrestErickson, 2025-01-19:
Regarding, "Ok I tested your file on freecad 0.21, "
Thanks for testing. 

My 0.21 has Python 3.8.10

I have Sheetmetal V0.7.3
[image]

I get errors when I open the file and switch to the work bench:

[image]

```
13:33:29  'type' object is not subscriptable
13:33:29  Traceback (most recent call last):
  File "<string>", line 68, in Initialize
  File "C:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Users\Admin\AppData\Roaming\FreeCAD\Mod\sheetmetal\.\SheetMetalUnfoldCmd.py", line 40, in <module>
    import SheetMetalNewUnfolder
  File "C:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Users\Admin\AppData\Roaming\FreeCAD\Mod\sheetmetal\.\SheetMetalNewUnfolder.py", line 479, in <module>
    class SketchExtraction:
  File "C:\Users\Admin\AppData\Roaming\FreeCAD\Mod\sheetmetal\.\SheetMetalNewUnfolder.py", line 484, in SketchExtraction
    edges: list[Part.Edge],

```

I have not yet found how to make an independent install of FreeCAD so that I can try a fresh install 0.21.

ForrestErickson, 2025-01-19:
Are you able to open my file in FreeCAD 1.0 with our errors?
Can you do so with out the migration dialog appearing every time you open the file?

If your system can migrate the file and you save it to a new name perhaps I could then open such a file.

shaise, 2025-01-19:
@ForrestErickson , With python 3.8 SheetMetal will work but without the new unfolder. If you want to work with the new unfolder under freecad 0.21 you can download the zip version of freecad 0.21 from here: https://github.com/FreeCAD/FreeCAD/releases/tag/0.21.2
unzip it to an empty folder and run freecad.exe from the bin folder.
You will need, again to uninstall/refresh cache/install sheetmetal

shaise, 2025-01-19:
If its ok for you to work with the old unfolder, you can go to:
`C:\Users\Admin\AppData\Roaming\FreeCAD\AdditionalPythonPackages\py38`
and delete the two folders starting with: networkx
This way you can use your original freecad 0.21 installation, you will just get the warning that the new unfolder is not available

ForrestErickson, 2025-01-19:
I am not using the unfolder (at least het) I want to get either my 0.21 or my 1.0 back where I can edit the file. 
## Notes 
I went to the folder and renamed the networkx folders to disable.
[image]

I opened 0.21 and opened the file. 
It had no reported errors when I switch to the work bench.  
[image]
I saved it as: ThreeU_Chassis_20250119_1358.FCStd

**This feels like progress.**

## Fixing Folds to Sketches
However How do I fix the folds back to the correct sketches?

For example FoldRear used to be driven by Sketch021
[image]

shaise, 2025-01-19:
I opened the file. Though the sketches are not under the fold in the tree, the fold still reference them:

[image]

shaise, 2025-01-19:
Ok - I found the reason. please wait - creating a new version

shaise, 2025-01-19:
Ok. new version available. If you use 0.21, when prompt for networkx installation, cancel it.
It should show the sketches correctly under the fold.

ForrestErickson, 2025-01-19:
I updated the Sheetmetal work bench.  
I opened the file.  
Sketches not in Folds.  
Lost of errors reported.    

[image]

```
14:56:19  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
<class 'ModuleNotFoundError'>: No module named 'SheetMetalFoldCmd'
14:56:19  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
<class 'ModuleNotFoundError'>: No module named 'SheetMetalFoldCmd'
14:56:19  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
<class 'ModuleNotFoundError'>: No module named 'SheetMetalFoldCmd'
14:56:19  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
<class 'ModuleNotFoundError'>: No module named 'SheetMetalFoldCmd'
14:56:21  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
<class 'ModuleNotFoundError'>: No module named 'SheetMetalFoldCmd'
14:56:21  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
<class 'ModuleNotFoundError'>: No module named 'SheetMetalFoldCmd'
14:56:21  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
<class 'ModuleNotFoundError'>: No module named 'SheetMetalFoldCmd'
14:56:21  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
<class 'ModuleNotFoundError'>: No module named 'SheetMetalFoldCmd'
```

ForrestErickson, 2025-01-19:
Disregard the previous post. I may not have installed correctly.  (Getting tired and confused).

[image]

ForrestErickson, 2025-01-19:
Although the install dialog terminates normally, the Sheetmetal work bench is not installing.  

[image]

I chose Ignore.

ForrestErickson, 2025-01-19:
Installed on FreeCAD 1.0 OK

[image]

## Progress with FreeCAD 1.0
[image]

# This Fixed It!
**Hurray!**

## An unrelated remaining problem, The Precomputation Migration Which Fails.
I assume this is NOT related to any work bench, however when I open the file I get this
[image]
I recomputed and save but get the same message.

Can you confirm you get the same message and cannot clear it?
File: [ThreeU_Chassis_20250119_1527.zip](https://github.com/user-attachments/files/18470327/ThreeU_Chassis_20250119_1527.zip)

I posted the earlier file to the FreeCAD Facebook two days ago and no one has tried to duplicate the problem with my file.

shaise, 2025-01-19:
What i meant is to use freecad 1.0 just to install sheetmetal. Now you can go back to 0.21 and your file should work without this error. (hopefully you have a copy NOT saved by 1.0)
The problem with this persistent error message in 1.0 when loading old file I encountered before.

ForrestErickson, 2025-01-19:
Regarding, "(hopefully you have a copy NOT saved by 1.0)"
I have a bewildering number of copies. Also I am a diligent user of Git and Github so as to have back copies.  It is the install of FreeCAD 0.21 and now 1.0 that are now unstable.

Anyway, thanks for your work bench and your work today.

Regarding, "... I encountered before."
It is frustrating that it was not mentioned by any of the many users on the FaceBook group before I stumbled into this nightmare.

##  Closing as fixed!!!

closed 2025-01-19 by ForrestErickson

shaise, 2025-01-20:
> ## Closing as fixed!!!

Hurray!