shaise/FreeCAD_SheetMetal #450: [NEW UNFOLDER] Issues with sketch and DXF on the new version.

thanhphong19512, 2025-01-20:
### Describe your issue

Hello, the latest new version works almost perfectly in many cases. However, there are still a few cases where I noticed issues. You can see them here:

**Case 1: Creating a sketch loses details (highlighted in the red box I marked).**

[image]

[image]

**Case 2: After exporting to DXF, if you re-import that DXF file into FreeCAD, you can see that it is not in the correct position.**

[image]

[image]

I’m attaching two files here for your reference.

[test.zip](https://github.com/user-attachments/files/18471701/test.zip)

### FreeCAD version info + SheetMetal WB version

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * sheetmetal 0.7.4
  * toSketch 1.0.1
```

shaise, 2025-01-20:
@alexneufeld , can you take a look?

thanhphong19512, 2025-01-20:
besides that the part that looks like the milling instead of sheet metal can still be unfolding

[image]

[image]

[Part4.zip](https://github.com/user-attachments/files/18479782/Part4.zip)

shaise, 2025-01-20:
> besides that the part that looks like the milling instead of sheet metal can still be unfolding
> 

That's a nice feature!!

alexneufeld, 2025-01-21:
> **Case 1: Creating a sketch loses details (highlighted in the red box I marked).**

case 1 is an issue with invalid shapes, or shapes with geometry with high tolerance values. The unfolded solid ends up having splits in the final solid, and segments get lost from the sketch profile. I can fix this by tracking the sketch edges during unfolding and adding an additional cleanup pass. This will take a few days to implement, please stand by for a fix.

> **Case 2: After exporting to DXF, if you re-import that DXF file into FreeCAD, you can see that it is not in the correct position.**

The `draft.makeSketch` function nullifies the sketch placement, including the rotation axis, then occasionally sets a spurious rotation angle, causing the final rotation in the DXF export to blow up due to floating point errors 🤦 . Fix by building the sketch ourselves. #451

thanhphong19512, 2025-01-23:
hello, can I keep this topic as a bug report?
I found new issue with this part. working on old unfolded but not the new one :D

new case:

[test 3.zip](https://github.com/user-attachments/files/18515947/test.3.zip)

Unfolded
[image]

Original
[image]

 Sketch export

[image]

PR #455 (2025-01-27): Improve unfold solid generation
add additional null-checks and cleanup passes, as well as constructing
the final solid from a single face, rather than using boolean operations.

closes #450

Extra details:

FreeCAD's (OpenCascade's?) tolerance requirements for shapes to pass a BOP check are that errors (misalignment between points, gaps in surface seams) are less than 0.1 NANOmeters. In my experience with test cases submitted by users in recent issues, as well as my own models, it's common for FreeCAD to produce geometry with errors on the order of 10-20 nm, which is much higher than the requi [... cut]
closed 2025-01-27 by shaise