shaise/FreeCAD_SheetMetal #453: Task view names Should be checked

FC-FBXL5, 2025-01-26:
Some naming of the task view panels is incorrect.

AddBase: **Generate Sheet Metal base shape**
AddWall: Flange parameters
Extrude: Extend parameters
AddFoldWall: Fold on sketch line parameters
Unfold: Unfold sheet metal object
AddCornerRelief: **Generate Sheet Metal base shape**
AddRelief: Add corner relief on solid
AddJuction: Add Junction **P**arameters
AddBend: Bend sharp corner **P**arameters
SketchOnSheet: -
AddCutout: Extruded Cutout **P**arameters
Forming: **Binded** faces/edges list
BaseShape: Generate Sheet Metal base shape

- 2x Copy/paste error: Generate Sheet Metal base shape
- Inconsistent use of capital letters: S/sheet M/metal, P/parameters
- Past tense of bind is bound.

shaise, 2025-01-26:
Thanks, I will take a look. 
What do you think is the proper casing of the title? only the first word or all?

FC-FBXL5, 2025-01-26:
I'm not a native English speaker, but as far as I know it should be:
First letter upper case and all following letters lower case, except nouns used as names should start with a capital letter, too.

Sheet Metal: refers to the **name** of the workbench.
sheet metal: refers to things made of sheet metal.
Unfold: refers to the name of the document object.
unfold: refers to a thing that is unfold or is used as a verb.

BTW - I would prefer SheetMetal for the workbench name like PartDesign, but there seems to be a tendency to split them into separate words, in the GUI and in the wiki.

shaise, 2025-01-26:
Sounds good to me.
I also like SheetMetal better, I will leave it to the DWG to decide.

shaise, 2025-01-26:
I have fixed all but SketchOnSheet and AddCutout sine I need to revise their task completely

FC-FBXL5, 2025-01-26:
I'm looking forward to the updated task panels. It is a big step forward to be able to (re-) select sketches in the task panel.

luzpaz, 2025-02-01:
@shaise close this ticket and open another to track `SketchOnSheet` and `AddCutout` ? Or do it from this one ?

shaise, 2025-02-02:
@luzpaz , this can be closed. The missing task panels are a work in progress, I'll just update my list

closed 2025-02-02 by shaise

luzpaz, 2025-02-02:
Is the list at issue 412 ?

shaise, 2025-02-03:
yes