shaise/FreeCAD_SheetMetal #454: Unfold Task view setting do not update properties immediately

FC-FBXL5, 2025-01-26:
I have found out that more options do not update the related properties directly:

Unfold Transparency has no effect until the OK Botton is pressed.

Color selectors have no direct effect, you have to uncheck and re-check either the Generate projection sketch option, the Separate projection layers option, or the Manual update object (might also require the Update button).

luzpaz, 2025-02-01:
Full [about](https://wiki.freecad.org/About) info (for posterity)

FC-FBXL5, 2025-02-02:
OS: macOS 15.2
Architecture: arm64
Version: 1.1.0dev.40041 (Git) Conda
Build type: Release
Branch: main
Hash: 64a4f6e3669279e5dbe9cb59d8e0123eaeb70833
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 72/133
Installed mods: 
  * Reinforcement
  * Movie 2025.1.4
  * freecad.gears 1.3.0
  * fasteners 0.5.36
  * Rocket 4.1.1
  * Ship 2024.11.26
  * sheetmetal 0.7.5
  * MakerWorkbench 1.0.1
  * Curves 0.6.54
  * Assembly3 0.12.2

Another finding: The color selectors for internal and bend lines are setting the other ones value.

shaise, 2025-02-02:
Some of the parameters are not updated immediately on purpose. Not all parameters are part of the object properties, and I did not want to do a recalculation every time a display property is changed.

FC-FBXL5, 2025-02-02:
> and I did not want to do a recalculation every time a display property is changed.

This seems to be a good idea, but it is one of the causes that I didn't realize that the **Bend line** color selector sets the **Unfold_Sketch_Internal**'s color properties and the **Internal color** selector sets the **Unfold_Sketch_Bends'** properties.

[image]

If a color is changed only the related sketches need to be updated not the unfold solid, and this happens in a fraction of a second.( In fact I don't know if FreeCAD allows the update of a single object.)

If I wanted to perform multiple changes between updates I could switch to manual update.

shaise, 2025-02-02:
I will take a look on it.