shaise/FreeCAD_SheetMetal #458: Productivity suggestions for mass production/ efficency.

ma00aem, 2025-02-01:
### Describe the feature

Hi! What a fantastic job you have made! I have some suggestions that would make this software not be up to paid software, but also exceeded what standard software delivers. 
 
* Different relief shapes

*Export options: -Export a whole assembly and auto generate all cut files. (This is such a time saver for all manufacturing people)
- File naming options. Custom with parameters. Example thickness, material, part id, quantity (this is also making production on a large scale easier)
- Layer naming and options for different types. (This is handy for making different layers for other software to automatically recognize a layer for example engraving layer
- Be able to make shortened bend lines custom sizing (can be used for example to engrave), This can be good if you have a laser cutter and want to engrave a line and not affect the material
- Auto add generated edge marks (marking with for example a small circle where the bend line is), is good for example when you plasma cut and don´t have something to engrave/etch with (i use this all the time)
- Be able to add text to the exported part for engraving (custom fields). This is great if you have a laser cutter and wants to have traceable parts

### Would you like to work on this issue?

None

luzpaz, 2025-02-01:
These types of tickets with multiple feature requests can lose their impact because of clutter and cross-talk regarding different issues. Would you consider splitting this ticket into several tickets ?

ma00aem, 2025-02-02:
See this for insperation 
https://apps.autodesk.com/INVNTOR/en/Detail/Index?id=7498862835390395853&appLang=en&os=Win64