shaise/FreeCAD_SheetMetal #459: Parameters sizing: Thickness, width and height. (remove if it already exist)

ma00aem, 2025-02-01:
### Describe the feature

This is so great if you could make that it automatically generate those to properties that later can be formatted to a partlist on a drawing. This is also soooo nice for speed and quality. Quality goes up when you don't have to manually enter size, where people have a tendency to be sloppier than a computer :D

Size can always be thickness and the sheet metal unfold width and height so use those 3 parameters.
The name can be different for different languages.

### Would you like to work on this issue?

None

shaise, 2025-02-02:
Hi @ma00aem ,

Sorry but I did not understand what exactly do you mean?

ma00aem, 2025-02-02:
In Inventor you have parameters and iProperties. 
If you can print out the parameter width , height and thickness then you can call it in the iProperties field by a combination of using text and calling for a parameter. If i example create i iProperty named dimension i can create it by using a forumla =<width> "x" <height> "x" <thickness> and get the result (width x height x thickness) for example 233x232x2

If a customer then want to have a sertain metadata i can create the cusomers dimension metadata by then linking yours and all works fine. I know that freecads parameter is not working excactly like this and the parameter thing in freecad is an own chapter for improvment :D

luzpaz, 2025-02-02:
It sounds like this is an [Expressions](https://wiki.freecad.org/Expressions) related feature request for the [Proerties tab](https://wiki.freecad.org/Property_editor), right? These are the areas you want to improve with your Feature Request for SheetMetal ?

ma00aem, 2025-02-02:
@luzpaz yes that is on point what i want. I want the dimension to be populated for width, height and thickness, and automatically update if there is a change. Then i can pick them up with the expression feature and make a combination if i understand you correctly. That would be great! (Maybe this allready exists?)

luzpaz, 2025-02-07:
@alexneufeld if you have a moment, can you weigh-in ? TIA

alexneufeld, 2025-02-08:
If you want to use the sheet thickness as a parameter in expressions, make the first feature of your sheetmetal part a BaseBend and call `BaseBend.Thickness` in the expression editor. for width and height of the unfolded shape, do `Unfold_Sketch.Shape.BoundBox.XLength` or `.YLength`. I don't think any new features are needed here.

[image]

ma00aem, 2025-02-14:
It would be good to have button to add some property so you can transform parts that are not made with basebend so it always can get those parameters. Because now you can sheet metals directly from a sketch and then you don´t get those parameters. Am i right?

alexneufeld, 2025-02-15:
If you have a solid sheet metal part, you always already know its thickness, either from a parameter of a dependent feature (such as the length of a pad) or the length of one of the solid's edges. "button to add some property" doesn't make sense, that would require a kind of reverse dependency injection that FreeCAD does not do. This issue should be closed.

ma00aem, 2026-09-21:
Can someone show me how to do this with a part and drawing example? This to me has been a quality improvement for my use with Autodesk Inventor as it has automatically populated the dimensions for my parts list. When you start working with drawings and you not have to fill in anything manually as it´s all information from the model is soooo great. Thanks /Andreas