shaise/FreeCAD_SheetMetal #460: Different relif shapes for corners.

ma00aem, 2025-02-01:
### Describe the feature

Different relief shapes for corners. There is a rounded relief shape that is really good for laser/plasma cutting

### Would you like to work on this issue?

None

luzpaz, 2025-02-01:
Here's a tip: if you make your proposal as elaborate as possible, there is a higher probability that it would be implemented. This Feature Request is very sparse.

shaise, 2025-02-02:
There is already a round relief shape to corners. can you attache images of the types you propose?

ma00aem, 2025-02-02:
Hi, excuse me for the lack of detail. I'm unsure if i can provide pictures from other softwares, so i tried to describe with words. Is it ok to make links to screenshots or oterwise i gladly can show you how it works in other software and give you a better understanding of what i try to describe.

shaise, 2025-02-02:
Link to screenshots is good enough

ma00aem, 2025-02-02:
https://help.autodesk.com/view/INVNTOR/2021/ITA/?guid=GUID-A7F8AB3C-A9AC-43AB-86FF-45133B124F7A
Se under Laser Weld Relief Shape

shaise, 2025-02-02:
Thank you, I understand now.

alexneufeld, 2025-02-02:
This documentation page has a better list of Inventors available relief shapes 
https://help.autodesk.com/view/INVNTOR/2021/ENU/?guid=GUID-EEFF7315-EF98-4922-AE22-09F9C6B3D488

Blendware, 2026-09-01:
Hi, if no one is currently working on this, I’d like to work on this feature.

shaise, 2026-09-02:
That will be great!

luzpaz, 2026-09-08:
@Blendware thank you. Hows the progress ?

Blendware, 2026-09-08:
@luzpaz  I think I should be done soon. I do push code regularly on my [fork](https://github.com/Blendware/FreeCAD_SheetMetal/tree/WeldCorner) if you'd like to take a look.

Currently I'm aware of 2 bugs:

1\) Unfold doesn't work on corner with size > 0 because refining geometry fails.

|Size = 0:|Size > 0 (current behavior)|Size > 0 (how it should be)|
|---|---|---|
|[image]|[image]|[image]

2\) Shape will fail if corner doesn't have gaps on both sides (I think I need to make it work if there is at least 1 gap)

|No gaps|With gaps|
|---|---|
|[image]|[image]|

luzpaz, 2026-09-16:
Nice work! Is there a reason to keep this ticket open ?

Blendware, 2026-09-20:
While the corner itself is rounded, we could argue that the transition to the fold is not.
If not, I'm for closing the ticket.

luzpaz, 2026-10-03:
> While the corner itself is rounded, we could argue that the transition to the fold is not.
If not, I'm for closing the ticket.

@shaise can you weigh-in? Thanks!

shaise, 2026-10-03:
I think @ma00aem should comment if the fix solved his request, but other than that I think this can be closed.

luzpaz, 2026-10-08:
Closing. Feel free to re-open if necessary.

closed 2026-10-08 by luzpaz