shaise/FreeCAD_SheetMetal #461: FreeCAD is unable to install the Python dependency for Sheetmetal (Networkx). When I launch FC, it shows "Networkx missing" error.

raindropsfromsky, 2025-02-02:
### Describe your issue

I followed these instructions.

1. Launched Addon Manager and uninstalled Sheetmetal.
2. Refreshed the local cache.
3. Installed Sheetmetal. It wants to install Pythin dependencies (Networkx). I clicked OK.
4. FC shows this error:
```
This addon requires Python packages that are not installed, and cannot be
installed automatically. To use this addon you must install the following Python
packages manually:
• networkx
To ignore this error and install anyway, press 0K.
```
5. I have no option but to press OK. FC installs Sheetmetal without Neetworkx.
6. When I restart FC, it shows this error:
```
Networkx dependency is missing and required for the new Unfolder. 
Try uninstalling Sheetmetal, refresh Addon Manager's cache, and reinstall
```

### FreeCAD version info + SheetMetal WB version

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40041 (Git) Conda
Build type: Release
Branch: main
Hash: 64a4f6e3669279e5dbe9cb59d8e0123eaeb70833
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/123.928
Installed mods: 
  * Beltrami 1.1.0
  * CommandPanel
  * CurvedShapes 1.0.13
  * Curves 0.6.54
  * ExplodedAssembly
  * fasteners 0.5.37
  * frame 0.1.1
  * FrameForge 0.1.2
  * FreeCAD-Ribbon 1.7.0
  * freecad.gears 1.3.0
  * FreeGrid 2.2.0
  * OpenTheme 2024.9.1
  * parts_library
  * Quetzal 1.1.0
  * SearchBar 1.3.3
  * sheetmetal 0.7.5
  * Silk 0.1.5
  * symbols_library
```

shaise, 2025-02-02:
This is an addon manager error, not sheetmetal. Sometimes the dependency servers are busy. Try waiting some time, uninstall and reinstall again.

raindropsfromsky, 2025-02-02:
I concede that this may not be an error from the addon.
But where should I escalate this issue to get it resolved?
Could you please help me route the complaint to the appropriate person?

If the server takes excessive time to respond, FC should wait and inform the user that the servers are busy, try after some time.
But in this case, the error says "cannot install automatically", which clearly means that the user should try a manual installation.

It is ok if a manual installation is indeed available.
In that case, FC should provide a link from where he can do the manual installation.

Clearly, FC does not handle this well.

Logically, someone should be in charge of this.

luzpaz, 2025-02-02:
Thanks for being persistent. You are correct that the Addon Manager needs loads more polish and development. 

>  Could you please help me route the complaint to the appropriate person?

This is an upstream FreeCAD main issue so it's more appropriate for https://github.com/FreeCAD/FreeCAD/issues

All currently open addon manager tickets: https://github.com/FreeCAD/FreeCAD/issues?q=is%3Aissue%20state%3Aopen%20label%3A%22Mod%3A%20Addon%20Manager%22

ovo-Tim, 2025-02-02:
Hi, you can install it manually, try this command: `pip install networkx --upgrade --target ~/.local/share/FreeCAD/AdditionalPythonPackages/py313/`

Please make sure your local python version is the same as the FreeCAD.

shaise, 2025-02-03:
for windows use 
`pip install networkx --upgrade --target C:/Users\<user name>\AppData\Roaming\FreeCAD\AdditionalPythonPackages\py311`
replace <user name> with the your windows username

shaise, 2025-02-03:
@Syres916 , nice!
perhaps we should embed this into the WB?

Syres916, 2025-02-03:
> perhaps we should embed this into the WB?

Absolutely no issue with you adding it, we may have to modify it in the future for Snap users, there's some issue with the actual pip that gets run, see https://github.com/FreeCAD/FreeCAD/pull/19384
It's entirely up to you but maybe wait until chennes PR gets merged and then see it it breaks my code.

raindropsfromsky, 2025-02-04:
> for windows use `pip install networkx --upgrade --target C:/Users\<user name>\AppData\Roaming\FreeCAD\AdditionalPythonPackages\py311` replace with the your windows username

While trying this solution, I found that pip command does not run in my laptop.
So I will have to install pip first. 

At this rate, the solution is getting out of my hand fast, as I am not familiar with Python system. 
Nor am I a coder. 

So all these steps are unwarranted work for users like me.
FC should handle all dependencies on its own, without expecting users to run from pillar to post.

raindropsfromsky, 2025-02-04:
FWITW I have raised this [bug](https://github.com/FreeCAD/FreeCAD/issues/19380) at FreeCAD Github.

luzpaz, 2025-02-06:
reposting:   
@raindropsfromsky the fix is in revision 40090 and currently the latest weekly is at r40077. New build will be triggered tomorrow.

shaise, 2025-02-12:
@Syres916 ,
Is this backward compatible with the previous addon manager version? (specifically 1.0 version)

Syres916, 2025-02-12:
Plenty of testing on Linux and Windows but I've no access to a Mac or any variety of Snap builds but here's a list that were all successful back to 0.21.1:

```
OS: Windows 7 build 7601
Architecture: x86_64
Version: 1.0.0rc4.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * A2plus 0.4.66
  * CfdOF 1.27.7
  * Curves 0.6.47
  * Dracula 0.0.5
  * fcSC-workbench
  * FreeCAD-themes 2024.9.14
  * freecad.gears 1.2.0
  * sheetmetal 0.6.0
```

```
OS: Windows 7 build 7601
Architecture: x86_64
Version: 1.1.0dev.39745 (Git)
Build type: Release
Branch: main
Hash: 6bb424b5d16adffb1896187a0725945f9635d144
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * A2plus 0.4.66
  * CfdOF 1.27.7
  * Curves 0.6.47
  * Dracula 0.0.5
  * fcSC-workbench
  * FreeCAD-themes 2024.9.14
  * freecad.gears 1.2.0
  * sheetmetal 0.6.0

```

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * sheetmetal 0.7.10
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.29.7

```

Upcoming version 1.0.1:

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39166 (Git)
Build type: Release
Branch: releases/FreeCAD-1-0
Hash: 630f7bcd2921aec689c1aa4c50bae19043e35076
Python 3.12.3, Qt 5.15.13, Coin 4.0.3, Vtk 9.1.0, OCC 7.8.2.dev
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * sheetmetal 0.7.10
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.29.7
```

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * sheetmetal 0.7.10
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.29.7

```

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.40206 (Git)
Build type: Release
Branch: main
Hash: 1350dd6bd0492cc52019f4c4e74203405e80cc29
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Logical/physical DPI: 96/93.1131
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * sheetmetal 0.7.10
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.29.7

```

Syres916, 2025-02-22:
@shaise the 1.1dev AddonManager (hopefully) has been fixed with regard to Snaps and AppImages ([commit](https://github.com/FreeCAD/FreeCAD/commit/f92fcbd089695c155c16bd0019f36544a78e2166)) and I've amended the macro above to include the Snap fix. The 1.0 branch should get the backport of the fix so in 1.0.1 this Python dependencies will be picked up correctly but there's always going to be a lag after release so it's probably a good idea to have this as an option available even if discretely in the menu somewhere.

shaise, 2025-02-23:
Thanks. I'm messing up with the code now quite a lot (fixing the tools to reduce core code, and adding some ui) Hopefully I'll have something stable soon

Syres916, 2025-02-26:
I'm glad I noted all the builds that I tested my code on because the latest 40408 is already been broken, too many changes going on at the moment.