shaise/FreeCAD_SheetMetal #462: Auto generate all cut files from assembly

ma00aem, 2025-02-02:
### Describe the feature

Export a whole assembly and auto generate all cut files. (This is such a time saver for all manufacturing people)
Instead of making one plate at a time this is a time saver.

### Would you like to work on this issue?

None

shaise, 2025-02-02:
This is a problematic because you have to select a face for each unfold. therefore it can not be done automatically.

luzpaz, 2025-02-02:
@alexneufeld any ideas for a workaround on how this could be automated ?

luzpaz, 2025-02-02:
@GS90 has some form of batch processing of sheetmetal parts in https://github.com/GS90/addFC?tab=readme-ov-file#overview  
Perhaps GS90 could weigh-in as well ?

GS90, 2025-02-02:
I believe that an automatic parts unfolding system should contain more than just "unfolding" and this task is more related to the bill of materials.

@ma00aem, you can try what is in [addFC](https://github.com/GS90/addFC/blob/main/repo/doc/quickstart_EN.md#basic-work-with-sheet-metal-parts-and-their-batch-processing), but at the moment, due to frequent changes in the new unfolding system, it does not work very stably...

ma00aem, 2025-02-03:
@GS90 addFC seam like a really handy tool! My own experience with using  a fantastic plugin called dxf exporter to Inventor is that it so easy to use. It will automatically create dxfs for all sheet metal parts when choosing an assembly and it is so fast! It has many really useful features with for example the ability to customize the file names (we use for example material, thickness, drawing number and pos & quantity as the file name, which make the next step for adding my parts for nesting really simple. The tool also have the ability to put custom names on the dwf (we use the drawings number and pos) . The ability to customize layer colors and layer name (also good for the next step of automatically interoperate a layer to for example engraving). It has a lot more useful features that could be interesting for you to see. Thanks for all the good work you guys do! I hope that i can help with inspiring you to keep going and make freecad a top tool! /Andreas

alexneufeld, 2025-02-03:
> [@alexneufeld](https://github.com/alexneufeld) any ideas for a workaround on how this could be automated ?

The *correct* way to do this is as follows:

- build each of the parts of your assembly as normal
- use links to add each part to an assembly
- create parametric unfolded versions of each of your parts using the FreeCAD Gui
- create a trivial python script that opens each FreeCAD file in your project folder and iterates over document objects, exporting every unfold sketch as a dxf. This script could be run using a make-file or other build system for publishing your project

Using separate files for each task (and cross-document links) or frozen sub-shape binders will improve your user experience during this process.

This is the same workflow you should be using to export CAM jobs, 3d printing models, FEA results, techdraw pages, etc. No complicated macros are required. Just create parametric objects as normal, then run one single line of python code (`ImportGui.somemethod` or whatever) to export each relevant document object.

The key insight here is to be explicit in what you want to export while building your project, rather than relying on a complex macro system that applies heuristics in order to create export data. The latter approach leads to buggy systems made of spaghetti code.

ma00aem, 2025-02-06:
We have a unfold script in Inventor that automatically make the unfold based on the largest surface.It orients the part one the largest straight surface. This works very well and we only needs to redefine the a-surface (how it's oriented) one a few occasions which is often when we have a strangely shaped plate when the drawing and the measurement in the part list is mismatched. 
So for me there is no need to manually select the unfold face. 
The tool we use to make the cut files then automatically exports the unfolded dxf of the assembly's all sheet metal parts. So there is no added value to have a new sketch inside of the part as long as it's exported if you know what i mean.
And i think how it works is that it can tell if a part is created with sheet metal or not. Then it takes all the sheet metal files and check if they already has a unfolded part. If not then it creates it. Then it export it and run through the presets of how different lines should be marked with color and layer names. Then sets the file name if the presets where you can choose which property that the file name should contain (for example material, thickness, drawing number and id). (The later steps with presets would be something to build on later if and not necessary to start with)