shaise/FreeCAD_SheetMetal #468: Parts broken when updating from Freecad 0.21.2 to 1.0.0

RexLinz, 2025-02-04:
### Describe your issue

I have several designs where I used SheetMetal 0.4.1 in FreeCAD 0.21.2. Trying to open these files in FreeCAD 1.0.0 **all** the designs are broken. One part out of a larger design in the files attached.

1. Open existing file in FreeCAD says "recomputation required", press YES
2. resulting in "Recompute error", report view displaying "PocketCoverScrews: SketchBased: No faces found in this direction"
3. Going back to the history BaseBend002 is fine, BentTop as well but starting with BaseBot the bends added are done on wrong faces resulting in more strange intermediate parts and finally the SketchScrewFront for PocketCoverScrews look like not linked to a valid face causing the final error message in the recomputation. See the screenshots included (looks like the order of the files got scrambled by the upload).

I tried several things to overcome with different results but without success:

- Upgrading to the latest version of SheetMetal 0.7.5 in FreeCAD 0.21.2 the part break at a different stage.
- Incremental upgrades of SheetMetal in FreeCAD 0.21.2, saving the design after each version step did break the part as well. I tried to step up using each first and last 0.x.y release (e.g. 0.4.1 -> 0.4.26 -> 0.5.0 -> 0.5.10 -> 0.6.0)
- Directly opening on FreeCAD 1.0.0 with SheetMetal 0.7.5 resulting in errors in recomputation and broken parts as well.

For me this look like being a combination of the topology problem with some other issues depending on the way I try to upgrade to the recent versions of SheetMetal + FreeCAD.

Is there any path to follow to get the parts to up to date file versions? Redoing the parts from scratch would be quite a lot of work and might break manufacturing data if I miss anything during that process.

Thanks for any help,
Roland

### FreeCAD version info + SheetMetal WB version

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Austria (de_AT)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.3
  * sheetmetal 0.4.1
  * toSketch 1.0.1
```

### Put here any screenshots or videos (optional)

original file using FreeCAD 0.21.2 and SheetMetal 0.4.1
[PowerBoxBase.zip](https://github.com/user-attachments/files/18657068/PowerBoxBase.zip)

recomputation required
[image]

failed to recompute
[image]

PocketCoverScrews broken
[image]

Preview still look ok!
[image]

Selecting individual steps in the component histors 
BendBot wrong position
[image]

Subsequent bends broken as well
[image]

GS90, 2025-02-04:
I encountered this problem myself, I'm staying on version 0.21.2.
At the moment there is no way to work correctly with sheet metal in version 1.0 and higher.
This is a topology [issue](https://github.com/FreeCAD/FreeCAD/issues/18671).

RexLinz, 2025-02-05:
Thanks for conforming it's not me doing something wrong. 

I did some more tests. From that maybe I should change the title of this issue to "parts broken when upgrading from SheetMetal 0.4.1 to any newer version".

As mentioned above (maybe not pointing out clear enough) the parts also break if I upgrade from SheetMetal 0.4.1 to 0.7.5 **within FreeCAD 0.21.2**. Even when doing the upgrade of SheetMetal incremental and saving the file after each upgrade step. Here my results when open the file just upgraded to SheetMetal 0.5.10 within FreeCAD 0.21.2. No error messages are displayed in that process at all.

1. Open file: no message to recompute, part seems OK. Manual recompute is not showing errors in that case.

[image]

2. Go down to BaseBend, open SketchBack and close the sketch. 
-> part flips to be bent backwards, SketchHolesBase is no longer on correct face, so holes from PocketHolesBase are broken as well. I have switched on display of this sketch to show the location after this step. It looks like the sketch is now on the wrong side of the base and also shifted down, maybe aligned with the bottom instead of top edge as external reference.

[image]

Note PocketCoverScrews, PocketScrewsSide are broken too as their sketches moved to wrong faces also.

Any ideas regarding this? At the moment it looks I'm locked to FreeCAD 0.21.2 with SheetMetal 0.4.1. 

Did I simply start with the wrong version to build my designs? For future compatibility at some point I will have to upgrade to a newer version, but it looks like I have bad luck and have to rebuild all my parts done so far when upgrading.

Thanks once more.
Roland

RexLinz, 2025-02-05:
Some more tests, I stepped thru versions of SheetMetal once more:

- Upgrading from SheetMetal 0.4.1 to 0.4.2 does not trigger the issue
- Upgrading to 0.4.3 the part is broken as shown above, even if saved after opening with 0.4.2 before.

Next I tried to fix the model. Going that way I have been able to upgrade my design to SheetMetal from 0.4.1 to 0.7.5 without intermediate steps

- Open design created in FreeCAD 0.21.2 using SheetMetal 0.4.1 with SheetMetal 0.7.5 installed.
- Change the edge the BaseBend is related to the correct one. This is throwing an recompute error, but will be fixed by the next step
- Updated all faces the following sketches for the pockets (holes) are linked to
- Saved as new file, reopened for test -> OK!
Just a sidenote: This did not help with FreeCAD 1.0.0. So at the moment I have to wait for this is fixed before doing any more designs in FreeCAD 1.0.0.

It still would be useful to find get a fix allowing to upgrade to a newer version of SheetMetal without this manual steps which might be quite a bit of work on larger designs.

Anyway, thanks for this great work, SheetMetal and Fasteners are installed with all my FreeCAD installations.

RexLinz, 2025-02-07:
Thanks for this comments, especially making clear the 4.2 to 4.3 changes will break existing designs. I'll have to rework all my existing designs using SheetMetal to step across the 4.2 to 4.3 change (upgrade to 0.7.5). Actually I'll have to do that in the main design, as there are external geometries being used I cut off for the example sent. I'll do on my notebook on a working copy of the designs and once it worked immediately upgrade my Desktop PC to SheetMetal 0.7.5 to get consistent.

Remapping the sketches is not so complicated if you know where they should be

1. In the part tree activate the last operation before the broken sketch
2. Select the face the sketch should be mapped to 
3. Use "Sketch / Map Sketch to face" from the main menu to remap the sketch
4. Repeat obove steps for any subsequent sketches broken (three in total in the example published).

I'm looking forward the topology issue being fixed, so finally I we all can step up to FreeCAD 1.0.0 (or whatever is the recent version then).

shaise, 2025-02-07:
Yes I used the sketch remap option, but the sketch came out rotated 180 degrees with part of the holes outside the face. I had to play with the mapping offsets and remove some reference geometry to make it work.

RexLinz, 2025-02-07:
Hello, 

I spent some time today to rework ALL my designs using SheetMetal upgrading to 0.7.5.

I checked your fix provided. In that 
•	SketchScrewsFront is linked to the inner side (I had that linked to the outside face)
•	SketchHolesBase is linked to the outside (I had that linked to the inside face)
•	SketchHolesSide is linked to the inner side (I had that linked to the outside face)

This explains why they did not match and some reference geometry might break. But obviously you can’t guess how I have done the design initially. Mapping to the correct faces the upgrade works without removing any reference geometry. For some of them I also had to correct direction of the bends (typically where I used the “reversed” flag).

Now I can upgrade to SheetMetal 0.7.5. on all my PC's 👍

luzpaz, 2025-02-07:
Can this be closed ?

RexLinz, 2025-02-08:
Last question left:

GS90 mentioned _"At the moment there is no way to work correctly with sheet metal in version 1.0 and higher"_. 
Is this generally true (so at the moment I have to avoid using 1.0.0 when planning to use SheetMetal in new designs) or just when upgrading existing files?

GS90, 2025-02-08:
Regarding work in version 1 and higher, there are #342 and #343.

RexLinz, 2025-02-08:
I read this can confirm it looks like at the moment SheetMetal look like not useful in 1.0.0 because of this topology issue.

closed 2025-02-08 by RexLinz

shaise, 2025-02-08:
There are some issues with topology on V1.0 on, but I did not notice it was worse then before when you start a new file. If you load and old one, then yes, sometimes the recalculation of the names breaks the model. I have also posted a [FreeCAD issue ](https://github.com/FreeCAD/FreeCAD/issues/18671)regarding problems with topology and external workbenches. These issues are not present on the RealThunder version.