shaise/FreeCAD_SheetMetal #469: Create Unfloding as new body

jaisekjames, 2025-02-05:
### Describe the feature

If we can generate auto unfolding in new Body with shape-binder, shape will  unfold automatically & we can keep adding new features.

[image]

**Note:** Sample file attached. Rename file [remove .zip]

[Auto unfolding.FCStd.zip](https://github.com/user-attachments/files/18669918/Auto.unfolding.FCStd.zip)

### Would you like to work on this issue?

None

jaisekjames, 2025-02-05:
Note: Change Set tip/Suppress features to see unfolding changes.

shaise, 2025-02-05:
You are correct! and this is on my list. I thought of making it as a simple object, not a body, though.
Did you make it automatically or manually?

jaisekjames, 2025-02-05:
I made it manually. When I saw Autodesk Inventor screenshot, they are using same tree structure [ like two separate body. This one rename to flat pattern.]

[image]

FC-FBXL5, 2025-02-08:
What about an automatic switch? 
If the object to unfold is inside a body, the Unfold solid and maybe the sketches are put into a new body for a PartDesign workflow, or else independent objects are created for a Part workflow.

shaise, 2025-02-08:
@FC-FBXL5 this was my initial thought, but putting it in another body with all the extras (such as the binder, origin and extra tree level), just for a single solid does seem to be a logical outcome.

Mannshoch, 2025-03-07:
In Solidworks the Unfold is a separate Child configuration (Update value with the mail body) which is usually used by Drawings. In model is not possible to have both bodies at the same time, but in a Drawing you could show different configuration per view.
Therefor you have the possibility to create several Unfoldings for the same body. e.g. 

- If you wish to show the folding process in several steps you could duplicate the configuration for as many steps you need.
(therefor every Unfold Feature in tree has per bow a child feature which could be deactivated) 
- You could visualize production steps like an unfolded and an unfolded and punched case

pierreporte, 2025-03-08:
I agree with @Mannshoch. The unfolded configuration should be some sort of state, not a new body.

shaise, 2025-03-08:
I agree as well, however since there is no such feature in part design, for the time being I want to have it as a separate object outside the body. (but parametric-ally linked)

pierreporte, 2025-03-08:
What feature is needed in Part Design ? We could open an issue to request it. I don’t think that it is variant parts, or more exactly not the kind of variant parts that @pieterhijma is doing. It’s something else that allow to have a part in undeformed state and deformed state, like hoses, cables, things like that.

shaise, 2025-03-08:
The issue is that Part Design holds a single solid. (designated as the tip)
If we can have a way to hold extra solids under another designation this will be great. Same as we can have extra sketches and linked objects that does not have to be necessarily a part of the object design.

pieterhijma, 2025-03-14:
> What feature is needed in Part Design ? We could open an issue to request it. I don’t think that it is variant parts, or more exactly not the kind of variant parts that [...]

I think this is exactly something that the variant parts could do. I'm not at all familiar with the sheet metal workbench, but hypothetically what I would like to have is an enum property "Folds" in the root of a sheet with an entry for each fold (fold1, fold2, or so).

Then when you want to show your model in one of the unfolded states, you select the fold you want to see and the model unfolds itself. 

However, it may also be useful to have your model in different states, one that shows the folded form, the other one the unfolded for plasma cutting or so. In that case,  you can (for example) create a separate file with a variant of the folded model and then switch to the enum case where everything is unfolded. You then have two instances of the same model in different states.

By the way, from what I understand in this issue with example above where folds are controlled by the Tip inside a body, the above is already possible in my implementation: It needs a property of type XLinkList or XSubLinkList with all the Bend objects, an enum property that serves as an index in this list, and then the tip is an expression in terms of the list and the enum.

belinski-roman, 2025-03-20:
> I agree as well, however since there is no such feature in part design, for the time being I want to have it as a separate object outside the body. (but parametric-ally linked)

#19735

belinski-roman, 2025-03-20:
Is there a problem with the Cyrillic alphabet
[image]

shaise, 2025-03-20:
hmmm, I wonder why. Python should handle it correctly.

shaise, 2025-03-22:
@belinski-roman , please uninstall/reinstall and check again.

belinski-roman, 2025-03-22:
> [@belinski-roman](https://github.com/belinski-roman) , please uninstall/reinstall and check again.

I saw that the problem was fixed, but reinstalling did not help. It's not critical yet and I'll wait for the next update.

belinski-roman, 2025-05-14:
> [@belinski-roman](https://github.com/belinski-roman), пожалуйста, удалите/переустановите и проверьте ещё раз.

The problem is still there

[image]