shaise/FreeCAD_SheetMetal #472: Delete edge a bend is related to from selection menu

RexLinz, 2025-02-11:
### Describe the feature

Reworking a couple of files from an SheetMetal 0.4.1 to 0.7.5 (using FreeCAD 0.22.1) I found a thing which could improve usability a lot. Due to topology problems the edge several bends has been attached moved and caused recompute errors. The bad thing has been to locate the edge to be able to unselect them, adding the correct edge instead.

**Sample to demonstrate the problem and suggested solution**

Part overview, edge to which the bend has been remapped due to FreeCAD's topology issue is just a 2mm long edge you won't recognize without scrolling and zooming each detail of the complete part. You just find the recompute error in the Report view and the exclamation mark (!) at the respective bend.

[image]

Detail of the edge causing the recompute error. Obviously it is not possible to add the bend there, so it's not a problem of SheetMetal.

[image]

From the flange parameters you find it is "Edge 70", but where it is located?

**Suggested improvment**
If I can mark the edge and use "delete" key in the Task view it would be easy to remove where ever it is in the design. Afterwards I can select the correct edge(s) to fix the topology. This is equal to the behavior of the task dialog box of Fillet and Chamfer in the Parts Design toolbox, both allow deleting edges from the Task view.

[image]

Unfortunately I never programmed in Python, so I even if I find the code in SheetMetal where this dialog box is defined I won't know how to add the ability to delete. From the Python console I guess one has to register a function to handle the delete event, as once you mark the line and press the delete key the console show:

> # Gui.runCommand('Std_Delete',0)

Thanks!

### Would you like to work on this issue?

None

shaise, 2025-02-12:
Yes, good idea!
Perhaps I will also add a "Delete selected" button. For more usability.

RexLinz, 2025-02-12:
Just another examples. 

a) Think about a bend gets remapped to an edge within a relief of a more complicated part. I think one might spend quite a time to find it on the structure. Any way to delete from the list of edges will be helpful.

[image]

b) A bend might get remapped to a non linear shape like an arc (not happened to me so far). This is resulting in a strange error message during recompute and the whole part disappears from the display. I have forced such in the file attached. Open the last bend in the part history and press select to view the part and edge it is related to.

[EdgeSelection_0.22.1.zip](https://github.com/user-attachments/files/18770136/EdgeSelection_0.22.1.zip)

[image]

shaise, 2025-02-12:
It seems not so easy to add this feature, as it is not so easy to know which item in the list belong to which edge in the drawing. However I at least added a "Clear selection" button that appear when you press the select button. This will clear all edges, and until I find a better solution, I hope this is good enough.

RexLinz, 2025-02-12:
Of course, marking the edge selected in the table with different color will not help in case the edge is well hidden.

The main use case will definitely be if FreeCAD scrambled the numbering of edge(s). Most likely not a single one of a bend with more edges attached will be correct in that case. And in many cases (at least in my designs) I just have a single edge. 
So clearing the whole list to start over easy is just fine to start over without searching the whole model to be able to deselect.

I already pulled the latest state of the repository, will test tomorrow. 
Thanks for this great work!

closed 2025-02-12 by RexLinz