shaise/FreeCAD_SheetMetal #479: `SheetMetal_AddBase` stopped working

pierreporte, 2025-03-08:
### Describe your issue

When using `SheetMetal_AddBase` with an existing valid sketch (open or closed), there is an error in the notification zone and a stacktrace in the report view. The feature is created in the tree bus nothing appears on the screen and it is marked as erroneous.

The error message and stacktrace are slightly different if the sketch is open or closed. “BaseBend: BRepOffsetAPI_MakeOffsetShape not done” with open sketches, “BaseBend: BRepSweep_Translation::Constructor” with closed sketches.

<details>
<summary>Closed sketch stacktrace</summary>

```
14:37:50  pyException: Traceback (most recent call last):
  File "/home/thomas/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalBaseCmd.py", line 164, in execute
    s = smBase(
        ^^^^^^^
  File "/home/thomas/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalBaseCmd.py", line 77, in smBase
    wallSolid = sketch_face.extrude(sketch_face.normalAt(0, 0) * thk)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'Part.OCCError'>: BRepSweep_Translation::Constructor
14:37:50  BaseBend: BRepSweep_Translation::Constructor
14:37:50  pyException: Traceback (most recent call last):
  File "/home/thomas/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalBaseCmd.py", line 164, in execute
    s = smBase(
        ^^^^^^^
  File "/home/thomas/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalBaseCmd.py", line 77, in smBase
    wallSolid = sketch_face.extrude(sketch_face.normalAt(0, 0) * thk)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'Part.OCCError'>: BRepSweep_Translation::Constructor
14:37:50  pyException: Traceback (most recent call last):
  File "/home/thomas/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalBaseCmd.py", line 164, in execute
    s = smBase(
        ^^^^^^^
  File "/home/thomas/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalBaseCmd.py", line 77, in smBase
    wallSolid = sketch_face.extrude(sketch_face.normalAt(0, 0) * thk)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'Part.OCCError'>: BRepSweep_Translation::Constructor
14:37:50  pyException: Traceback (most recent call last):
  File "/home/thomas/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalBaseCmd.py", line 164, in execute
    s = smBase(
        ^^^^^^^
  File "/home/thomas/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalBaseCmd.py", line 77, in smBase
    wallSolid = sketch_face.extrude(sketch_face.normalAt(0, 0) * thk)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'Part.OCCError'>: BRepSweep_Translation::Constructor
```
</details>

<details>
<summary>Open sketch stacktrace</summary>

```
14:35:54  pyException: Traceback (most recent call last):
  File "/home/thomas/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalBaseCmd.py", line 164, in execute
    s = smBase(
        ^^^^^^^
  File "/home/thomas/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalBaseCmd.py", line 77, in smBase
    wallSolid = sketch_face.extrude(sketch_face.normalAt(0, 0) * thk)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'Part.OCCError'>: BRepSweep_Translation::Constructor
14:35:54  BaseBend: BRepSweep_Translation::Constructor
14:35:54  pyException: Traceback (most recent call last):
  File "/home/thomas/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalBaseCmd.py", line 164, in execute
    s = smBase(
        ^^^^^^^
  File "/home/thomas/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalBaseCmd.py", line 77, in smBase
    wallSolid = sketch_face.extrude(sketch_face.normalAt(0, 0) * thk)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'Part.OCCError'>: BRepSweep_Translation::Constructor
14:35:54  pyException: Traceback (most recent call last):
  File "/home/thomas/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalBaseCmd.py", line 164, in execute
    s = smBase(
        ^^^^^^^
  File "/home/thomas/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalBaseCmd.py", line 77, in smBase
    wallSolid = sketch_face.extrude(sketch_face.normalAt(0, 0) * thk)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'Part.OCCError'>: BRepSweep_Translation::Constructor
14:35:54  pyException: Traceback (most recent call last):
  File "/home/thomas/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalBaseCmd.py", line 164, in execute
    s = smBase(
        ^^^^^^^
  File "/home/thomas/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalBaseCmd.py", line 77, in smBase
    wallSolid = sketch_face.extrude(sketch_face.normalAt(0, 0) * thk)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'Part.OCCError'>: BRepSweep_Translation::Constructor
```
</details>

### FreeCAD version info + SheetMetal WB version

```shell
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40408 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b07caa732e63064b8851ecdda184292962d95c15
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96.0236/117.596
Installed mods: 
  * CosmeticThread3D
  * sheetmetal.backup1731333427.8204525 0.5.3 (Disabled)
  * Curves 0.6.59
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.21
```

shaise, 2025-03-08:
Can you add an example file?

pierreporte, 2025-03-08:
@shaise [sheetmetal_issue_479.zip](https://github.com/user-attachments/files/19145222/sheetmetal_issue_479.zip) contains the FCStd file. It has two sketches, one closed and one open. When one is selected and the command activated, the errors I mentioned are displayed.

shaise, 2025-03-09:
It works great on my version:
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40444 (Git) Conda
Build type: Release
Branch: main
Hash: 089605bb4db98668661bf56db3f5f5bb4f7f0d87
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical/physical DPI: 96/90.0633
Installed mods: 
  * fasteners 0.5.37
  * sheetmetal_25-02-24.7z
```

Let me try the latest FC

pierreporte, 2025-03-09:
Just downloaded the latest weekly. Same behavior.

```
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96.0236/117.596
Installed mods: 
  * CosmeticThread3D
  * sheetmetal.backup1731333427.8204525 0.5.3 (Disabled)
  * Curves 0.6.59
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.21
```

shaise, 2025-03-09:
I will try to get a hold on a Linux machine.
Do you know if it is the same when using the 1.0 official release?

pierreporte, 2025-03-09:
Same issue with FreeCAD 1.0. I temporarily removed the local settings folder so it’s unlikely to be linked to that.

```
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * sheetmetal 0.7.21
```

pierreporte, 2025-03-09:
It must be OCC version: I just tested on Windows with 1.0 and it worked. It has OCC 7.8.1 like you. It’s strange that Linux version ship an older OCC.

shaise, 2025-03-09:
Tried on Ubuntu and works:

[image]

```
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/90.0633
Installed mods: 
  * fasteners 0.5.21
  * sheetmetal 0.7.21
```
This is almost exact same configuration, so I'm truly puzzled...

shaise, 2025-03-10:
So it the same OS now and exactly the same FreeCAD ver (and OCC) so clearly its not the OCC version. I even tried changing language to French, but still, no error messages.
I really wonder whats the difference.

pierreporte, 2025-03-10:
I made a full reset of FreeCAD but kept the workbenches, and it works. I haven’t touched the workbench settings though, just updated it. It’s not the first time I run into issues after updating. I thought there was an issue about this last time (the python dependency was missing) but no trace of it.

FC-FBXL5, 2025-03-11:
BTW - No problem on macOS 

```
OS: macOS 15.3.1
Architecture: arm64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 72/133
Installed mods: 
  * Reinforcement
  * Movie 2025.1.4
  * freecad.gears 1.3.0
  * fasteners 0.5.37
  * Rocket 4.1.1
  * Ship 2024.11.26
  * Cables 0.1.3
  * sheetmetal 0.7.21
  * MakerWorkbench 1.0.1
  * Curves 0.6.58
  * Assembly3 0.12.2
```