shaise/FreeCAD_SheetMetal #480: Features need geometry to be selected before activation

pierreporte, 2025-03-08:
### Describe your issue

Except `SheetMetal_BaseShape`, the workbench requires the user to first select geometry to apply features on them. It is impractical, especially because it’s no clear from start what needs to be selected. Sure, there are the tooltips, but it is also inconsistent with core FreeCAD started to move towards allowing starting features before geometry selection, and less necessary since the introduction of task panels.

This issue is created as a bug because it could be considered a UX bug.

### FreeCAD version info + SheetMetal WB version

```shell
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40408 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b07caa732e63064b8851ecdda184292962d95c15
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell /.7.0, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96.0236/117.596
Installed mods: 
  * CosmeticThread3D
  * sheetmetal.backup1731333427.8204525 0.5.3 (Disabled)
  * Curves 0.6.59
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.21
```

shaise, 2025-03-08:
I was thinking the exact same, when I added the task panels. Now that there is a task panel for each command, this is indeed possible. This is not so trivial as I need to handle bad selections. I need to find a way to disable rendering of the feature + disable the OK button of the task until a valid selections is made.
Added to my list

luzpaz, 2026-02-28:
@alexneufeld any interest in tackling this ?