shaise/FreeCAD_SheetMetal #481: The workbench lacks a sketch creation command

pierreporte, 2025-03-08:
### Describe the feature

The sheet metal workbench works with sketches, but it doesn’t offer a command to create them. Users need to either go to PartDesign or Sketcher workbenches to do so. It would be a good idea to provide one of the two commands, `Sketcher_NewSketch` or `PartDesign_NewSketch` (preferably the PD version). Once https://github.com/FreeCAD/FreeCAD/issues/12106 is resolved, the new unique command could replace it.

### Would you like to work on this issue?

None

shaise, 2025-03-08:
Agree completely. Added to my list.

zokhasan, 2025-07-04:
https://github.com/user-attachments/assets/acd18903-9f9f-4d99-92a1-4e21a6c52f28

shaise, 2025-07-04:
anyone knows how to do it by code?

davesrocketshop, 2025-07-04:
Import the Sketcher commands

`import SketcherGui`

Add it to your toolbar (example from the Rocket WB)

`self.appendToolbar(translate('Rocket', 'Rocket'),
                        ['Separator', 'Rocket_NewSketch', 'Sketcher_EditSketch', 'Separator', 'Rocket_Calculators'])`

zokhasan, 2025-07-08:
I was able to do the same as in the video (like "Customize…" command) but through code.
What other buttons should be on this extra panel by default? Maybe "Create body", "Create group"... or only "Create sketch"?

zokhasan, 2025-07-09:
I mean, for SheetMetal workbench it might be enough to add only "create sketch" from Sketcher workbench..

shaise, 2025-07-09:
exactly

zokhasan, 2025-07-09:
@shaise create new branch for it and I will suggest adding to it. In theory everything shoold work fine, but it's better to test it in a separete branch before mergin it to the main code branch.

zokhasan, 2025-08-26:
@luzpaz , I already did it. You can see the solution I proposed at the link: https://github.com/shaise/FreeCAD_SheetMetal/pull/503