shaise/FreeCAD_SheetMetal #482: `SheetMetal_Extrude` is limited and not intuitive

pierreporte, 2025-03-09:
### Describe the feature

The `SheetMetal_Extrude` command is used for two things:

1. Extend an existing flange by offsetting the edge (with some parameters).
2. Add material at the end of a flange following a sketch that MUST touch another flange, that would be cut to make room for the tab.

While use 1 is fairly intuitive, use 2 is harder to use and the user must read the wiki in order to understand how to make the sketch and know that it needs to span across two flange. Also, having to use the property view when there is already a task panel is definitely not user-friendly.

I propose splitting the feature in two new features: `SheetMetal_Extend` and `SheetMetal_AddTab`. `SheetMetal_Extrude` should be kept for backwards compatibility but not shown to users.

`SheetMetal_Extend` would be for use 1, and a basic copy of the direct-access version of `SheetMetal_Extrude`, complete with task panel and options.

`SheetMetal_AddTab` would be used for use 2 and more or less be the opposite of `SheetMetal_SketchOnSheet`. Its task panel would wait for just a sketch to create more material. There could be options to optionally create automatic cutouts when the sketch touches other flanges, to reproduce current feature set. If the sketch is created from a flange before a bend, it could make a cutout below the new flat flange.

### Would you like to work on this issue?

None

pierreporte, 2025-03-24:
@shaise What’s your opinion on this?

shaise, 2025-03-27:
Hi @pierreporte , sorry for the late reply.
Frankly, I totally forgot use case 2 feature exists. I tend to agree it should be a separate feature. 
I will add it to my list.

shaise, 2026-02-04:
@pierreporte ,
I have revised the code such that extending by sketch will only require selecting the sketch. (The sketch must be attached to the flange to be extended)
On the way I refactored the code so it can handle more complex scenarios, specially when clearance is needed to clear collisions.  

Before:  

[image]

[image]

After:  

[image]

[image]
  
shai.