shaise/FreeCAD_SheetMetal #483: Making folded walls with sketches

pierreporte, 2025-03-09:
### Describe the feature

PR #467 introduced the ability to make a folded flange along an existing plane or surface, but it only allows to make them rectangular. If the user wants the outer edge to have a specific shape, it is necessary to make one or more cutouts.

`SheetMetal_AddWall` or a new feature could provide the ability to select an existing sketch, of which at least one line is within the sheet thickness, so that the resulting wall follows its contour. Holes should be supported, but only if they don’t touch the thickness. If they touch the bend, then the bend is cut.

The line that lies withing the sheet thickness could be required to have a more specific position but I can’t really make a choice right now. It would be best if we discuss it.

### Would you like to work on this issue?

None

pierreporte, 2025-03-10:
@sheetmetalman This could interest you.

sheetmetalman, 2025-03-10:
@pierreporte Indeed! Very nice!

But I'm currently in college, so I won't be able to develop any code in FreeCAD. I'll have to wait for vacations in July or December or even the next year vacations.

Straight away, what I can say now is the following:
1. The first thing to implement that is the ability to have one object (under the parent object of the AddWall command) to each wall/flange (with his own parameters, like radius and angle).
2. After that, we'll need to have a sketch that is the wall (but without thickness or radius) for each child object. And this step, I think, is the hard one.

sheetmetalman, 2026-03-26:
Hi, @pierreporte!

Sorry for the delay!

I would be happy to help, but I don't have time right now. I began a new role at work. And I am currently studying mechanical engineering at university. So I'm very busy!

sheetmetalman, 2026-03-27:
Thanks, @pierreporte ! By the way, congratulations on developing the new hem feature!