shaise/FreeCAD_SheetMetal #485: Missing File 'WrappedCutoutPanel.ui'

contradict, 2025-03-18:
### Describe your issue

When editing a part made with a previous version of this tool, i get the following traceback:
```
Traceback (most recent call last):
  File "/home/user/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalTools.py", line 661, in setEdit
    taskd = self.getTaskPanel(vobj.Object)
  File "/home/user/.local/share/FreeCAD/Mod/sheetmetal/./SketchOnSheetMetalCmd.py", line 248, in getTaskPanel
    return SMWrappedCutoutTaskPanel(obj)
  File "/home/user/.local/share/FreeCAD/Mod/sheetmetal/./SketchOnSheetMetalCmd.py", line 258, in __init__
    self.form = SheetMetalTools.taskLoadUI("WrappedCutoutPanel.ui")
  File "/home/user/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalTools.py", line 541, in taskLoadUI
    return Gui.PySideUic.loadUi(path)
  File "<string>", line 4, in <module>
<class 'RuntimeError'>: Cannot open file
```

### FreeCAD version info + SheetMetal WB version

```shell
OS: Arch Linux (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.2, Qt 6.8.2, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * Alternate_OpenSCAD 1.0.0
  * sheetmetal.backup1742265860.5223427 0.6.0 (Disabled)
  * sheetmetal 0.7.21
```

shaise, 2025-03-18:
Sorry... Thanks for the catch!
should be fixed now

contradict, 2025-03-18:
Thanks for fixing that so quickly. This is a nice tool, thank you!

closed 2025-03-20 by luzpaz