shaise/FreeCAD_SheetMetal #507: FreeCAD 1.0.1 and Sheet Metal

HOC-Vo, 2025-07-24:
### Describe your issue

Hello, I have a problem with version 1.0.1.

Last week, I designed a sheet metal part in Sheet Metal using version 1.0.0.
Today, I installed version 1.0.1 and can no longer edit the BaseShape in Sheet Metal.

The properties window no longer appears when I double-click, and a message appears in the output window.

The icon next to BaseShape also looks like a normal blue cube.

For a small project, Sheet Metal works satisfactorily as usual.

I have already installed version 1.0.1 and also Sheet Metal a second time - unfortunately, without success :?:

Cu60_Verbinder.FCStd is correct
Cu60_NH_L.FCStd is defect in version 1.0.1

### FreeCAD version info + SheetMetal WB version

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * OpenTheme 2025.5.20
  * sheetmetal 0.7.24
```

### Put here any screenshots or videos (optional)

![Image](https://github.com/user-attachments/assets/5abd0ec5-b852-47b6-8888-e3a0ff49

[FreeCAD.zip](https://github.com/user-attachments/files/21413041/FreeCAD.zip)

95f0)

pilatomic, 2025-07-30:
mpilato-fpsas here, but on my personal account.

I apologize for the incomplete report, I did not have much time to create the report from my workplace.
I have reproduced the issue at home, on a setup similar to the one I have at work : 

Freecad Version info : 
OS: KDE Flatpak runtime (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 878f0b8)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.12.10, Qt 6.9.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/unset/fusion
Installed mods: 
  * sheetmetal 0.7.24

Simple project that shows the issue, with the sketch line extended beyond the part to be folded :
[Freecad_Sheet_Fold_Issue.FCStd.zip](https://github.com/user-attachments/files/21516199/Freecad_Sheet_Fold_Issue.FCStd.zip)

And the same exception :
`20:45:09  pyException: Traceback (most recent call last):
  File "/home/pila/.var/app/org.freecad.FreeCAD/data/FreeCAD/Mod/sheetmetal/./SheetMetalFoldCmd.py", line 276, in execute
    s = smFold(
        ^^^^^^^
  File "/home/pila/.var/app/org.freecad.FreeCAD/data/FreeCAD/Mod/sheetmetal/./SheetMetalFoldCmd.py", line 107, in smFold
    CutSolids = FoldShape.cut(toolExtr)
                ^^^^^^^^^^^^^^^^^^^^^^^
<class 'ValueError'>: {'sclassname': 'N4Part18NullShapeExceptionE', 'sErrMsg': 'Null shape', 'sfile': '/run/build/FreeCAD/src/Mod/Part/App/TopoShapeExpansion.cpp', 'iline': 1395, 'sfunction': 'Part::TopoShape& Part::TopoShape::makeShapeWithElementMap(const TopoDS_Shape&, const Mapper&, const std::vector<Part::TopoShape>&, const char*)', 'swhat': 'Null shape', 'btranslatable': False, 'breported': False}`

I cannot exclude a case of user error from me, as I'm quite new to this workbench, but I just can't see where I could be going wrong.

luzpaz, 2025-08-01:
@shaise any suggestions for a better title for this ticket ?

shaise, 2025-08-01:
Perhapse: Unclear error message when invalid sketch used in AddFoldWall,