shaise/FreeCAD_SheetMetal #523: «Make Wall» error

GS90, 2025-09-04:
### Describe your issue

The latest update (0.7.50) breaks the "Make Wall" tool, with the values ​​"Material Inside" and "Thickness Outside" an extra element appears at the junction of the bends.

The error occurs in any version of FC.

### FreeCAD version info + SheetMetal WB version

```shell

```

### Put here any screenshots or videos (optional)

[image]

GS90, 2025-09-04:
> can you attach the file?

Yes, of course.
Attached in the main message.

shaise, 2025-09-04:
I have added a quick fix for this issue that still works with #521. However it will not work with some combinations, so there is still some work

GS90, 2025-09-04:
Thank you.

I checked on my models and did not find any serious errors.

The only thing that changed is the behavior in certain cases... Previously, when the values of "**reliefd**" and "**reliefw**" were both zero, that part (the red rectangle) was always cut off regardless of the "**Bend Type**" parameter.

Can this behavior be returned (at least for the sake of backward compatibility) or is the new one more correct?

[image]

[SM_Part.zip](https://github.com/user-attachments/files/22161606/SM_Part.zip)

shaise, 2025-09-05:
This was actually an error before. see #521. While trying to fix it i messed up with the corner case. when material inside is selected, and gaps are set, you don't want the bend to shorten un-bended metal as well. 
There are still problematic cases when 2 adjacent edges are bent, material inside and gaps i need to figure out how to fix.

[image]  [image]

shaise, 2025-09-06:
It should behave better now with a mix of gaps and inner material

GS90, 2025-09-09:
[image]
[image]

After upgrading to V0.7.52, I noticed a much bigger problem on the working parts... all sketches with projected geometry from the bends turned out to be inverted.
This is a very big big sadness for the already created parts :(

The error can be reproduced by opening and rebuilding before and after the fix (.51 and .52) the attached file.

[FD_Frame_01_Left.zip](https://github.com/user-attachments/files/22242923/FD_Frame_01_Left.zip)

GS90, 2025-10-19:
I tested it, it got much better.
Thank you for your work.

closed 2025-10-19 by GS90