shaise/FreeCAD_SheetMetal #525: <class 'AttributeError'>: 'NoneType' object has no attribute 'getContents' on using material spreadsheet for K-Factor

LightTemplar, 2025-09-04:
### Describe your issue

Using material spreadsheet for K-Factor throws following error:
```
18:03:10  pyException: Traceback (most recent call last):
  File "/home/user/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalUnfoldCmd.py", line 284, in execute
    shape, sketches = self.newUnfolder(fp, baseObj, baseFace)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalUnfoldCmd.py", line 225, in newUnfolder
    bac = BendAllowanceCalculator.from_spreadsheet(sheet)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalNewUnfolder.py", line 700, in from_spreadsheet
    r_t_header = sheet.getContents("A1")
                 ^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getContents'
18:03:10  Body001_Unfold: 'NoneType' object has no attribute 'getContents'
```

[Unnamed2.zip](https://github.com/user-attachments/files/22142658/Unnamed2.zip)

### FreeCAD version info + SheetMetal WB version

```shell
OS: Ubuntu 22.04.5 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: ProDark.qss/ProDark/Fusion
Installed mods: 
  * FreeCAD-themes 2025.6.4
  * CubeMenu
  * WebTools
  * OpenTheme 2025.5.20
  * ExtremeProDark 2.6.4
  * sheetmetal 0.7.50
  * 3D_Printing_Tools
  * ProDarkThemePreferencePack 1.0.0
  * WB_Organizer 2024.1.29
  * Manipulator 1.6.0
```

shaise, 2025-09-06:
Thanks for reporting!
It is indeed a bug and should be fixed now on ver V0.7.51.

LightTemplar, 2025-09-07:
And it works, thanks!

closed 2025-09-07 by LightTemplar