shaise/FreeCAD_SheetMetal #537: Unfold broken when part contains splines

kotarou3, 2025-09-28:
### Describe your issue

See the below screenshots and error messages.
Note that the examples also show issue #536 and use the same CAD file

CAD file: [spline-pocket.zip](https://github.com/user-attachments/files/22580406/spline-pocket.zip)

### FreeCAD version info + SheetMetal WB version

```shell
OS: Debian GNU/Linux 13 (trixie) (GNOME-Flashback:GNOME/gnome-flashback-metacity/xcb)
Architecture: x86_64
Version: 1.0.0.
Build type: Release
Python 3.13.5, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * ExplodedAssembly
  * sheetmetal 0.7.55
```

### Put here any screenshots or videos (optional)

Attempting to unfold using this face results in an "Intersection cannot be computed" error:
[image]
```
15:29:54  pyException: Traceback (most recent call last):
  File "/home/kota/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalUnfoldCmd.py", line 287, in execute
    shape, sketches = self.newUnfolder(fp, baseObj, baseFace)
  File "/home/kota/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalUnfoldCmd.py", line 229, in newUnfolder
    sel_face, unfolded_shape, bend_lines, root_normal = SheetMetalNewUnfolder.getUnfold(
  File "/home/kota/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalNewUnfolder.py", line 1342, in getUnfold
    sketch_wirelist = Edge2DCleanup.clean_and_structure_geometry(sketch_lines)
  File "/home/kota/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalNewUnfolder.py", line 994, in clean_and_structure_geometry
    intermediate_result1 = Edge2DCleanup.eliminate_bsplines(edges, spline2arc_tol)
  File "/home/kota/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalNewUnfolder.py", line 844, in eliminate_bsplines
    new_edge, max_err = Edge2DCleanup.bspline_to_arc(edge)
  File "/home/kota/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalNewUnfolder.py", line 804, in bspline_to_arc
    arc = Part.Arc(point1, point2, point3).toShape().Edges[0]
<class 'Part.OCCError'>: Intersection cannot be computed
15:29:54  Body_Unfold: Intersection cannot be computed
```

Attempting to unfold using this face results in the spline being "straightened out":
[image]
[image]

shaise, 2025-09-28:
@alexneufeld , any thoughts?

luzpaz, 2026-01-04:
@alexneufeld soft bump

alexneufeld, 2026-01-05:
> [@alexneufeld](https://github.com/alexneufeld) soft bump

most of the open issues related to unfolding at the moment are a result of opencascades lack of robustness when handling cylindrical faces with b-Spline edges. I'm working on refactoring the unfolding process to mitigate most of the known issues.

If you want to pass something along to the OCCT devs, ask them why the `toBiArcs()` method of the bspline edge in this step file fails miserably:

[image]

[biarc_issue.zip](https://github.com/user-attachments/files/24426433/biarc_issue.zip)

luzpaz, 2026-01-05:
> If you want to pass something along to the OCCT devs, ask them why the toBiArcs() method of the bspline edge in this step file fails miserably:

CC @dpasukhi (would you have a moment to weigh-in ?)